espressif/esp-matter #1473: Not able to commission Light Switch on Amazon Alexa

CodeIsaacAir, 2025-06-25:
**Describe the bug**
Been trying to add a light switch implementation(done on ESP32-C3) on my Amazon Alexa home but the following error keeps popping up. I've not modified the example either 

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id: v5.5
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
- 

[image]

- Host Machine OS: MacOS
- Host Machine Python version: Python 3.13.1

- Commissioner app and versions if present: Alexa
- Home hub app and versions if present: Echo Dot
- Commissioner's logs if present:

**Any additional details**
...

Jerry-ESP, 2025-06-25:
[image]

Amazon haven't supported light switch device yet.

CodeIsaacAir, 2025-06-25:
can you please provide the source of this image. would like to go through the entire list

Jerry-ESP, 2025-06-25:
This is the link, but you need a csa account: https://groups.csa-iot.org/wg/matter-wg/document/folder/2998

jonsmirl, 2025-07-02:
There is no point in adding a light switch device to Amazon/Google/Apple. They will likely never add them. Light switches are meant to be bound to lights, they don't generate anything Amazon/Google/Apple can receive.

What you may be looking for is a Generic Switch which can be commissioned to  Amazon/Google/Apple.

Jerry-ESP, 2025-08-04:
Hi @0xdharanesh ,

For your request, maybe you can try onoff plug device type: https://github.com/espressif/esp-matter/blob/ec9699be7c8e5ff43420107e7236c7a33953b452/components/esp_matter/data_model/esp_matter_endpoint.cpp#L483

0xdharanesh, 2025-08-06:
@Jerry-ESP, we have already used that for the socket-based. How can I differentiate the switch, then?

Jerry-ESP, 2025-08-06:
The Switch device type is a client device, it can send control command to other devices, but can't be controlled by App or controller.

0xdharanesh, 2025-08-11:
@Jerry-ESP, can I add it as a virtual endpoint?

Jerry-ESP, 2025-08-11:
Of course you can add this endpoint, it was implemented in the matter sdk, but not supported by mainly eco-systems yet.