espressif/esp-matter #1475: Failing to update the attribute `OperationalState::Attributes::CurrentPhase`

lboue, 2025-06-28:
**Describe the question/query that you have**

Hi,

I'm trying to update the attribute `OperationalState::Attributes::CurrentPhase` but so far without success.
Could you help me identify the problem with this attribute?

```c++
void DishwasherManager::StartProgram()
{
    mTimeRemaining = 30;
    uint32_t cluster_id = OperationalState::Id;
    UpdateOperationState(OperationalStateEnum::kRunning);
    UpdateDishwasherDisplay();

    ESP_LOGI(TAG, "Update OperationalState::Attributes::CurrentPhase");
    esp_matter_attr_val_t phase = esp_matter_int(1);
    esp_matter::attribute::update(1, cluster_id, OperationalState::Attributes::CurrentPhase::Id, &phase);
}

```

**Additional context**
...

lboue, 2025-06-30:
I don't want to update it on the client side. I want to update the value on the server side (esp-matter).

jadhavrohit924, 2025-06-30:
OperationalState asks for a delegate implemented by an application and esp-matter doesn't manage this attribute, so it is applications responsibility to update the storage.

lboue, 2025-06-30:
Thanks for the explanation. I'll look into it.