espressif/esp-matter #1479: ColorControl::ColorMode attribute is created without value specified in config

olavt, 2025-07-04:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: c7dd22d
- ESP-IDF Commit Id: v5.4.1
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6

There is a bug in extended_color_light::create.

This is my code:

```cpp
    extended_color_light::config_t light_config;
    light_config.on_off_lighting.start_up_on_off = nullptr; // Retain previous state
    light_config.level_control_lighting.start_up_current_level = nullptr;   // Retain previous value
    light_config.color_control.color_mode = (uint8_t)ColorControl::ColorMode::kCurrentHueAndCurrentSaturation;
    light_config.color_control.enhanced_color_mode = (uint8_t)ColorControl::ColorMode::kCurrentHueAndCurrentSaturation;

    ESP_LOGI(TAG, "CreateEndpoint: light_config.color_control.color_mode = %d", static_cast<int>(light_config.color_control.color_mode));
    ESP_LOGI(TAG, "CreateEndpoint: light_config.color_control.enhanced_color_mode = %d", static_cast<int>(light_config.color_control.enhanced_color_mode));

    // Create Extended Color Light Endpoint
    ESP_LOGI(TAG, "Creating extended color light endpoint");
    endpoint_t* endpoint = extended_color_light::create(matterNode->GetNode(), &light_config, ENDPOINT_FLAG_NONE, nullptr);
    if (endpoint == nullptr)
    {
        ESP_LOGE(TAG, "CreateEndpoint: Failed to create extended color light endpoint");
        return nullptr;
    }

    ESP_LOGI(TAG, "CreateEndpoint: Created extended color light endpoint");
```

The problem is that the ColorControl::ColorMode attribute is created with an incorrect value.

From the code it seems that the ColorMode attribute is created first with value ColorMode::kColorTemperature for some reason. Then the code tries to create it again with the color_control config (which specifies ColorMode::kCurrentHueAndCurrentSaturation). The second create of the attribute fails and there is an error message in the log:

I (592) MatterExtendedColorLight: CreateEndpoint: light_config.color_control.color_mode = 0
I (602) MatterExtendedColorLight: CreateEndpoint: light_config.color_control.enhanced_color_mode = 0
I (602) MatterExtendedColorLight: Creating extended color light endpoint
W (612) esp_matter_core: Attribute 0x00000002 on cluster 0x00000300 already exists. Not creating again.
I (622) MatterExtendedColorLight: CreateEndpoint: Created extended color light endpoint
I (632) MatterExtendedColorLight: CreateEndpoint: ColorMode is 2

The result is that the value for ColorMode in the config is not taken into consideration when the attribute is created.

jadhavrohit924, 2025-07-15:
`0x00fa` is not any random value, it was defined in the matter spec. If you want any other default value, you can always change it in the application. 
`light_config.color_control_color_temperature.startup_color_temperature_mireds = nullptr;`

Now the spec has updated we will check what can be done in such cases.

olavt, 2025-07-15:
@jadhavrohit924 Where in the spec was it specified that a default value for StartUpColorTemperatureMireds should be applied when the ColorMode is set to kCurrentHueAndCurrentSaturation?

I don't think it's reasonable to have such side effects and that one need to investigate the source code in order to understand that not setting a value for StartUpColorTemperatureMireds will force ColorMode to kColorTemperature.

jadhavrohit924, 2025-07-17:
@olavt Right, we will come up with solution on it soon.