espressif/esp-matter #1483: TC-CADMIN-1.15: mdns service not removed upon deleting fabric

mikesolsen, 2025-07-07:
**Describe the bug**
Test Case TC-CADMIN-1.15 creates three fabrics and then removes the second fabric. The corresponding mdns service for the second fabric is not removed as required by Step 10.

**Environment**
- ESP-Matter Commit Id: f353b9e3d400ad80b254c3594e57154f8fa32ee9
- ESP-IDF Commit Id: v5.4.1 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):

- Host Machine OS: n/a to this issue
- Host Machine Python version: n/a to this issue

- Commissioner app and versions if present: n/a
- Home hub app and versions if present: n/a
- Commissioner's logs if present:

**Any additional details**
It appears that DnssdServer::StartServer(Dnssd::CommissioningMode mode) in the connectedhomeip submodule is expecting all matter related services to be removed, and then it adds a new copy of services as required.

`EspDnssdRemoveServices()` is called within `ESP32DnssdImpl.cpp`, which calls `mdns_service_remove()`. A somewhat recent change to the mdns service has changed the behavior of `mdns_service_remove_for_host()`.  Only a single service is removed, when the previous implementation would remove all matching services. Please see esp-protocols commit link below.  I confirmed the new behavior by inserting additional log statements.

https://github.com/espressif/esp-protocols/commit/25b3d5fd7b1d42ba52c8faf819080ebec9af860b

I am using esp-protocols commit b8cdd373f47ca70b894c2d8a84456463b7c2252d, however I don't see any relevant changes to master.

DejinChen, 2025-07-08:
Thanks for the report! Had re-produced the issue, and it was caused by the commit you mentioned.

DejinChen, 2025-07-17:
We have added the fix to matter side and backported it to v1.4 branch https://github.com/project-chip/connectedhomeip/pull/39984
Will fix this issue in next submodule update.