espressif/esp-matter #1484: libble_app.a needed by 'room_air_conditioner.elf', missing and no known rule to make it

mimou-77, 2025-07-09:
**Describe the bug**
when building the example room_air_conditioner of esp-matter for the target esp32c6, i get this error:
ninja: error: 'esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/libble_app.a', needed by 'room_air_conditioner.elf', missing and no known rule to make it

Bug fixed by copying  ~/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6/libble_app.a into ~/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib

**Environment**
- ESP-Matter Commit Id: cb216647d443444af8cc858a009228849e70287f
- ESP-IDF Commit Id: 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): esp32c6

- Host Machine OS: Linux Ubuntu
- Host Machine Python version: Python 3.12.3

mimou-77, 2025-07-09:
Bug fixed by copying ~/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6/libble_app.a into ~/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib

shripad621git, 2025-07-10:
@mimou-77 , instead you can apply the changes made [here](https://github.com/project-chip/connectedhomeip/commit/0c90d7299f7ab70b325e8a7febd0f210ae629ce4).

dhrishi, 2025-07-16:
@mimou-77 The fix should be present in the latest main branch of esp-matter that supports new ESP-IDF. Please close the issue if it helps