espressif/esp-matter #1489: TC-PS-2.1 Test Harness Client reads ActiveBatChargeFaults from Server DUT Failing

bilalahmaddev, 2025-07-15:
Test logs:

INFO       | 2025-07-15 07:22:34.641929 | Executing Test Step: Step 32: Test Harness Client reads ActiveBatChargeFaults from Server DUT
CHIPTOOL   | 2025-07-15 07:22:35.002555 | CHIP:TOO Command: powersource read active-bat-charge-faults 0x66bbe46c14aa9fdb 19 
CHIPTOOL   | 2025-07-15 07:22:35.003321 | CHIP:TOO Sending command to node 0x66bbe46c14aa9fdb
CHIPTOOL   | 2025-07-15 07:22:35.003829 | CHIP:CSM FindOrEstablishSession: PeerId = [1:66BBE46C14AA9FDB]
CHIPTOOL   | 2025-07-15 07:22:35.004277 | CHIP:CSM FindOrEstablishSession: No existing OperationalSessionSetup instance found
CHIPTOOL   | 2025-07-15 07:22:35.005142 | CHIP:DIS Found an existing secure session to [1:66BBE46C14AA9FDB]!
CHIPTOOL   | 2025-07-15 07:22:35.005576 | CHIP:DIS OperationalSessionSetup[1:66BBE46C14AA9FDB]: State change 1 --> 5
CHIPTOOL   | 2025-07-15 07:22:35.006002 | CHIP:TOO Sending ReadAttribute to:
CHIPTOOL   | 2025-07-15 07:22:35.006408 | CHIP:TOO 	cluster 0x0000_002F, attribute: 0x0000_001E, endpoint 19
CHIPTOOL   | 2025-07-15 07:22:35.006813 | CHIP:DMG SendReadRequest ReadClient[0xffff7c02bfc0]: Sending Read Request
CHIPTOOL   | 2025-07-15 07:22:35.007207 | CHIP:DMG 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
CHIPTOOL   | 2025-07-15 07:22:35.007808 | CHIP:DMG >> to UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8]:5540 | 95730674 | [Interaction Model  (1) / Read Request (0x02) / Session = 49474 / Exchange = 36078]
CHIPTOOL   | 2025-07-15 07:22:35.008233 | CHIP:DMG Header Flags =
CHIPTOOL   | 2025-07-15 07:22:35.008630 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.010453 | CHIP:DMG     Exchange (0x05) =
CHIPTOOL   | 2025-07-15 07:22:35.010944 | CHIP:DMG     {
CHIPTOOL   | 2025-07-15 07:22:35.011380 | CHIP:DMG         Initiator = true
CHIPTOOL   | 2025-07-15 07:22:35.011808 | CHIP:DMG         NeedsAck = true
CHIPTOOL   | 2025-07-15 07:22:35.012362 | CHIP:DMG     }
CHIPTOOL   | 2025-07-15 07:22:35.012923 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.013205 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.013471 | CHIP:DMG Decrypted Payload (21 bytes) =
CHIPTOOL   | 2025-07-15 07:22:35.013744 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.014337 | CHIP:DMG     data = 1536001724021324032f24041e1818290324ff0b18
CHIPTOOL   | 2025-07-15 07:22:35.014628 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.014913 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.015271 | CHIP:DMG ReadRequestMessage =
CHIPTOOL   | 2025-07-15 07:22:35.015549 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.015822 | CHIP:DMG 	AttributePathIBs =
CHIPTOOL   | 2025-07-15 07:22:35.016087 | CHIP:DMG 	[
CHIPTOOL   | 2025-07-15 07:22:35.016489 | CHIP:DMG 		AttributePathIB =
CHIPTOOL   | 2025-07-15 07:22:35.016767 | CHIP:DMG 		{
CHIPTOOL   | 2025-07-15 07:22:35.017031 | CHIP:DMG 			Endpoint = 0x13,
CHIPTOOL   | 2025-07-15 07:22:35.017288 | CHIP:DMG 			Cluster = 0x2f,
CHIPTOOL   | 2025-07-15 07:22:35.017543 | CHIP:DMG 			Attribute = 0x0000_001E,
CHIPTOOL   | 2025-07-15 07:22:35.018050 | CHIP:DMG 		}
CHIPTOOL   | 2025-07-15 07:22:35.018351 | CHIP:DMG 			
CHIPTOOL   | 2025-07-15 07:22:35.018618 | CHIP:DMG 	],
CHIPTOOL   | 2025-07-15 07:22:35.018966 | CHIP:DMG 		
CHIPTOOL   | 2025-07-15 07:22:35.019246 | CHIP:DMG 	isFabricFiltered = true, 
CHIPTOOL   | 2025-07-15 07:22:35.019512 | CHIP:DMG 	InteractionModelRevision = 11
CHIPTOOL   | 2025-07-15 07:22:35.019780 | CHIP:DMG },
CHIPTOOL   | 2025-07-15 07:22:35.020045 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.020308 | CHIP:DMG Additional Fields =
CHIPTOOL   | 2025-07-15 07:22:35.020563 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.020826 | CHIP:DMG     peer_address = UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8]:5540
CHIPTOOL   | 2025-07-15 07:22:35.021090 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.021347 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.021602 | CHIP:EM <<< [E:36078i S:35643 M:95730674] (S) Msg TX from 000000000001B669 to 1:66BBE46C14AA9FDB [9087] [UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8]:5540] --- Type 0001:02 (IM:ReadRequest) (B:51)
CHIPTOOL   | 2025-07-15 07:22:35.021874 | CHIP:EM ??1 [E:36078i S:35643 M:95730674] (S) Msg Retransmission to 1:66BBE46C14AA9FDB in 333ms [State:Active II:500 AI:300 AT:4000]
CHIPTOOL   | 2025-07-15 07:22:35.022137 | CHIP:DMG MoveToState ReadClient[0xffff7c02bfc0]: Moving to [AwaitingIn]
CHIPTOOL   | 2025-07-15 07:22:35.022392 | CHIP:DMG << from UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8%eth0]:5540 | 136633768 | [Interaction Model  (1) / Report Data (0x05) / Session = 35643 / Exchange = 36078]
CHIPTOOL   | 2025-07-15 07:22:35.022651 | CHIP:DMG Header Flags =
CHIPTOOL   | 2025-07-15 07:22:35.022922 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.023177 | CHIP:DMG     Exchange (0x06) =
CHIPTOOL   | 2025-07-15 07:22:35.023433 | CHIP:DMG     {
CHIPTOOL   | 2025-07-15 07:22:35.023687 | CHIP:DMG         AckMsg = 95730674
CHIPTOOL   | 2025-07-15 07:22:35.024092 | CHIP:DMG         NeedsAck = true
CHIPTOOL   | 2025-07-15 07:22:35.024355 | CHIP:DMG     }
CHIPTOOL   | 2025-07-15 07:22:35.024819 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.025130 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.025408 | CHIP:DMG Decrypted Payload (33 bytes) =
CHIPTOOL   | 2025-07-15 07:22:35.025675 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.025947 | CHIP:DMG     data = 153601153500370024021324032f24041e18350124000118181818290424ff0c18
CHIPTOOL   | 2025-07-15 07:22:35.026206 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.026631 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.026914 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.027175 | CHIP:DMG Additional Fields =
[... 46 lines cut ...]
CHIPTOOL   | 2025-07-15 07:22:35.046014 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.046270 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.046523 | CHIP:DMG Additional Fields =
CHIPTOOL   | 2025-07-15 07:22:35.047346 | CHIP:DMG {
CHIPTOOL   | 2025-07-15 07:22:35.047636 | CHIP:DMG     peer_address = UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8]:5540
CHIPTOOL   | 2025-07-15 07:22:35.047913 | CHIP:DMG }
CHIPTOOL   | 2025-07-15 07:22:35.048168 | CHIP:DMG  
CHIPTOOL   | 2025-07-15 07:22:35.050856 | CHIP:EM <<< [E:36078i S:35643 M:95730675 (Ack:136633768)] (S) Msg TX from 000000000001B669 to 1:66BBE46C14AA9FDB [9087] [UDP:[2400:adc5:123:3b00:da3b:daff:fe9f:20d8]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
CHIPTOOL   | 2025-07-15 07:22:35.051179 | CHIP:EM Flushed pending ack for MessageCounter:136633768 on exchange 36078i
WARNING    | 2025-07-15 07:22:35.051989 | Test Failure: The test expects no error but the "FAILURE" error occured.

[TC-PS-2-1_2025_07_15_07_21_58.log](https://github.com/user-attachments/files/21231755/TC-PS-2-1_2025_07_15_07_21_58.log)

Device logs:

I (57209) chip[EM]: >>> [E:35203r S:42630 M:125590963] (S) Msg RX from 1:000000000001B669 [787A] to AB7D19ADD6E1758D --- Type 0001:02 (IM:ReadRequest) (B:51)
I (57219) esp_matter_attribute: ********** R : Endpoint 0x0013's Cluster 0x0000002F's Attribute 0x0000001E is <invalid type: 4> **********
E (57229) chip[DMG]: Attribute type 0x48 not handled
E (57229) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_002F, attributeId: 0x0000_001Eerr = 501
I (57249) chip[EM]: <<< [E:35203r S:42630 M:234137019 (Ack:125590963)] (S) Msg TX from AB7D19ADD6E1758D to 1:000000000001B669 [787A] [UDP:[2400:ADC5:**Environment**
- 
- 
- ESP-Matter Commit Id: matter 1.4
- ESP-IDF Commit Id: 5.41
- SoC (eg: ESP32 or ESP32-C3): esp32-s3

jadhavrohit924, 2025-07-17:
@bilalahmaddev Power source cluster should have delegate interface for such attributes, which is missing in the core cluster implementation. You can have a workaround as suggested by @Jerry-ESP, just make sure that you populate your ActiveBatChargeFaults value instead of empty list.

bilalahmaddev, 2025-07-17:
@Jerry-ESP Sure I will try. @jadhavrohit924 Power source cluster does not have implementation to set user defined delegate for optional attributes of PS cluster. Can you please point me to delegate in power  source cluster if there is any?

bilalahmaddev, 2025-07-18:
@Jerry-ESP Test case is now passing with your suggested change. Thanks

closed 2025-07-21 by dhrishi