espressif/esp-matter #1492: Failed to read attribute value in device management(DMG)

lunDreame, 2025-07-16:
**Environment**
- ESP-Matter Commit Id: 26f4e4d3f4e684edd68c353446c3fcc3ee901771 (release/v1.4)
- ESP-IDF Commit Id: f5c3654a1c2d2a01f7f67def7a0dc48e691f63c0 (v5.4.2)
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):

```log
E (6725) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6725) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6735) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6735) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6735) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6735) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
E (6735) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0039, attributeId: 0x0000_0011err = 57f
```

- Host Machine OS: macOS 15.6
- Host Machine Python version: 3.13.5

- Commissioner app and versions if present: 
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
Hello, Matter is currently experiencing an error like the device log attached above. We are using Matter Bridge. Therefore, we create a bridge_node endpoint and navigate to the respective endpoint information and find the BridgedDeviceBasicInformation cluster for that endpoint. And if the cluster exists, we create an attribute. NodeLabel is a rough flow. Why do I get the above error? Please let me know if you need more information.

wqx6, 2025-07-17:
Hi, @lunDreame 0x57f means UnsupportedEndpoint. Did you enable the bridged endpoint after creating it?

lunDreame, 2025-07-17:
Yes, that's right.