espressif/esp-matter #1493: TC-IDM-6.2 Event Subscription Access Control Issues

mikesolsen, 2025-07-16:
**Describe the bug**
Steps 14 and 16 of TC-IDM-6.2 fail.
Step 14: Should fail with UNSUPPORTED_ACCESS, but instead INVALID_ACTION is returned.
Step 16: Should succeed with no events reported. Instead INVALID_ACTION is returned.

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/f353b9e3d400ad80b254c3594e57154f8fa32ee9
- ESP-IDF Commit Id:  v5.4.1 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): 

[TC-IDM-6.2 Step 14 DUT logs.txt](https://github.com/user-attachments/files/21267286/TC-IDM-6.2.Step.14.DUT.logs.txt)
[TC-IDM-6.2 Step 14 TH Logs.txt](https://github.com/user-attachments/files/21267283/TC-IDM-6.2.Step.14.TH.Logs.txt)
[TC-IDM-6.2 Step 16 DUT Logs.txt](https://github.com/user-attachments/files/21267284/TC-IDM-6.2.Step.16.DUT.Logs.txt)
[TC-IDM-6.2 Step 16 TH Logs.txt](https://github.com/user-attachments/files/21267285/TC-IDM-6.2.Step.16.TH.Logs.txt)

- Host Machine OS: n/a
- Host Machine Python version: n/a

- Commissioner app and versions if present: n/a
- Home hub app and versions if present: n/a
- Commissioner's logs if present: n/a

**Any additional details**
DUT was factory reset just before rerunning the problematic test steps, to verify the issue reproducible and to capture clean logs.  Both Test Harness and DUT logs have been attached for Steps 14 & 16.

Jerry-ESP, 2025-07-17:
This is a known issue and the test plan should be changed: 
https://github.com/project-chip/connectedhomeip/issues/29203 
https://github.com/CHIP-Specifications/chip-test-plans/issues/3442

mikesolsen, 2025-07-17:
Excellent.

closed 2025-07-17 by mikesolsen