espressif/esp-matter #1494: TC-IDM-10.2: Descriptive cluster Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features:

bilalahmaddev, 2025-07-17:
TC-IDM-10.2: Descriptive cluster Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 

```
MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.WARNING
    test_name: Spec XML parsing
    location: 
       Endpoint: 0,
       Cluster:  67 (0x43) ProxyDiscovery
       Command:  1 (0x01)
    problem: Command with unknown direction
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.WARNING
    test_name: Parse Device Type XML
    location: 
       DeviceType: 0
    problem: Device type Introduction does not have an ID listed
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 1,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 2,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 3,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 4,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 5,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 6,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 7,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 8,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 9,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.061 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 10,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 11,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 12,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 13,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 14,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 15,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 16,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 18,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 19,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 

[MatterTest] 07-17 06:24:41.062 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
       Endpoint: 20,
       Cluster:  29 (0x1d) Descriptor
      Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: TAGLIST, implemented features: 
    spec_location: 
```

[TC_IDM_10_2_2025_07_16_20_10_36.log](https://github.com/user-attachments/files/21284411/TC_IDM_10_2_2025_07_16_20_10_36.log)

ESP-Matter Commit Id: matter 1.4
ESP-IDF Commit Id: 5.41
SoC (eg: ESP32 or ESP32-C3): esp32-s3

Jerry-ESP, 2025-07-17:
If you want to add the tag_list attribute, you should add the TAGLIST feature. 
https://github.com/espressif/esp-matter/blob/882edef5a08ff75c6aa11f14d6de14ab1520255a/components/esp_matter/esp_matter_feature.cpp#L59-L82

Jerry-ESP, 2025-07-17:
So what is your TH version?