espressif/esp-matter #1495: Can the Controller and Bridge coexist in the project?

0x1abin, 2025-07-17:
I see that CONFIG_ESP_MATTER_ENABLE_MATTER_SERVER and CONFIG_ESP_MATTER_COMMISSIONER_ENABLE are mutually exclusive. I want to make sure that it is possible for the Matter Controller and Bridge to run on the same ESP32S3 at the same time.
https://github.com/espressif/esp-matter/blob/release/v1.4/examples/controller/main/matter_project_config.h#L5-L15

Jerry-ESP, 2025-07-18:
Do you want to enable commissioner in the controller? And add the bridge at the same time?

0x1abin, 2025-07-19:
I need to enable Controller, Commissioner and Bridge at the same time.

Jerry-ESP, 2025-07-21:
The Bridge is a Matter server device, which requires Matter commissioning.
The Commissioner, on the other hand, is a client device that performs commissioning for other devices.

If you place both the bridge and the commissioner on the same device, it becomes impossible to commission the bridge itself, because a device cannot act as both commissioner and commissionee in the same commissioning flow.
Therefore, it is generally not recommended to implement both a commissioner and a Matter server on the same device.

However, combining a controller and a bridge is acceptable and does not pose such issues.

Additionally, most customers do not use our chip to build a commissioner product.
Here’s a simple question to consider: How would you scan the QR code?

0x1abin, 2025-07-22:
I want to implement fabric-bridge application. My device can be used as an independent controller to commission and control other 
Matter devices, and can also add commissioned devices to the internal Bridge and share commissioning to other Matter ecosystems.
On Linux, the controller and bridge are usually two independent processes. Is it possible to run on the same device on esp-matter?

https://project-chip.github.io/connectedhomeip-doc/examples/fabric-bridge-app/linux/README.html
https://project-chip.github.io/connectedhomeip-doc/examples/fabric-admin/README.html

ph2355, 2026-01-07:
Has anyone been able to achieve this? I too am trying to accomplish bridge + commissioner combo. 
@Jerry-ESP Regarding the issue that running the same device as a commissioner and commissionee: surely there is a workaround for this, maybe hardcoding the fabric table or modifying the commissioning flow in esp-matter/connectedhomeip.

@jonsmirl esp with external PSRAM may be able to handle this, but I fear you would need to rewrite a big part of esp-matter or connectedhomeip since commissioner + bridge combo is mutually exclusive. I don't quite understand what you mean by implementing the bridge server clusters in the controllers Matter context. The way you describe, there would be a commissioner/controller which would have bridge clusters, but that would mean that the bridge would be unable to be commissioned to any other fabric, correct? (since the device is a commissioner, not commissionee) 

The functionality I would like to achieve is to have a commissioner that can commission external devices and the bridge as a commissionee that can be commissioned into other fabrics. The bridge should be automatically part of the commissioners fabric. The commissioner and commissionee functionality does not have to work at the same time.