espressif/esp-matter #1497: How to add HoldTime, HoldTimeLimit attributes to Occupancy Sensing cluster?

olavt, 2025-07-19:
I'm creating an Occupancy Sensor endpoint using this code:

```cpp
    occupancy_sensor::config_t occupancy_sensor_config;

    // Set the Occupancy Sensing Features
    if (occupancySensor->SupportsSensingTechnology(OccupancySensor::SensingTechnology::Radar)) {
        occupancy_sensor_config.occupancy_sensing.feature_flags |= cluster::occupancy_sensing::feature::radar::get_id();
    }
    
    endpoint_t* endpoint = occupancy_sensor::create(matterNode->GetNode(), &occupancy_sensor_config, ENDPOINT_FLAG_NONE, nullptr);
```

Then I want to add the optional attributes HoldTime and HoldTimeLimits.

HoldTime seems quite straight forward to add with this code:

```cpp
    // Create the optional attributes for the Occupancy Sensing cluster
    cluster_t *occupancy_sensing_cluster = cluster::get(endpoint, OccupancySensing::Id);
    esp_matter::cluster::occupancy_sensing::attribute::create_hold_time(occupancy_sensing_cluster, 4);
```

If I test this without adding the HoldTimeLimits, I get this error when trying to access the HoldTime attribute from chip-tool:

[1752949785.769] [1086:1086] [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)

Is the reason for getting this error that the HoldTimeLimits attribute have not been added?

How do I add the HoldTimeLimits attribute?

The create_hold_time_limits function looks like this:

```cpp
attribute_t *create_hold_time_limits(cluster_t *cluster, uint8_t* value, uint16_t length, uint16_t count)
{
    return esp_matter::attribute::create(cluster, OccupancySensing::Attributes::HoldTimeLimits::Id,
                                         ATTRIBUTE_FLAG_NONE | ATTRIBUTE_FLAG_MANAGED_INTERNALLY, esp_matter_array(value, length, count));
}
```

In the Matter specification, the HoldTimeLimits should be a structure like this:

[image]

But, the create_hold_time_limits does not take this kind of structure as a parameter.

Any example code for this?

Jerry-ESP, 2025-07-21:
Please refer to this example: https://github.com/project-chip/connectedhomeip/blob/252a32ba39f8497bf9406aab810bb4d8a77fd75c/examples/all-clusters-app/all-clusters-common/src/occupancy-sensing-stub.cpp#L43-L76

You need to call SetHoldTimeLimits() to init the HoldTimeLimits attribute. similar to the taglist attribute implement in esp-matter: https://github.com/espressif/esp-matter/blob/882edef5a08ff75c6aa11f14d6de14ab1520255a/examples/generic_switch/main/app_main.cpp#L246

olavt, 2025-07-22:
@Jerry-ESP Ok, I thought the parameters to esp_matter::cluster::occupancy_sensing::attribute::::create_hold_time_limits would init the attribute.

I think the parameters to esp_matter::cluster::occupancy_sensing::attribute::::create_hold_time_limits seem a bit strange. What are the parameters?