espressif/esp-matter #1507: TC-RR-1.1: Performance issue with many bridged group clusters

mikesolsen, 2025-07-28:
**Describe the bug**
TC-RR-1.1 will fail in Step 14 "Setting group key map" if the DUT contains a large number of group clusters.  The request to set the group key map will quickly timeout in the TH before the DUT can finish processing it.

**Environment**
- ESP-Matter Commit Id: 40ca7ea9c3ae485ea678a3a40676a2915f8bbe0d (july 24)
- ESP-IDF Commit Id: v5.4.1 4c2820d377d1375e787bcef612f0c32c1427d183
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): attached below.

- Host Machine OS: Ubuntu 24.04.2 LTS
- Host Machine Python version: 3.12.3

- Commissioner app and versions if present: Version: v2.11+fall2024
- Home hub app and versions if present: n/a
- Commissioner's logs if present: attached below.

**Any additional details**
I am implementing a Matter Bridge. Each bridged endpoint includes a group cluster. I am testing with a large number of devices to evaluate performance and determine how large the NVS data partition needs to be.

The TC-RR-1.1 Step 14 creates 1 group key for each possible group (4 groups per group cluster). The TH appears to send group keys in batches of 40. The processing time to append a single group key escalates quickly, resulting in a test failure.

I made a couple small modifications to test the behavior:
1) added logs to `WriteGroupKeyMap()` in `group-key-mgmt-server.cpp`. The addition of these logs does not meaningfully change the outcome.
      `WriteGroupKeyMap: remove all` is logged before calling `provider->RemoveGroupKeys()`
      `WriteGroupKeyMap: Append <index>` is logged before calling `provider->SetGroupKeyAt()` under `ListOperation::AppendItem`
2) Added `interactionTimeoutMS = 64000` to the `client.WriteAttribute` call in Step 14 of `TC_RR_1_1.py`.

Appending the first group key on the first fabric: 70ms
Appending the second group key on the first fabric: 130ms
Appending the 41st group key on the first fabric: 240ms
Appending the 320th group key on the first fabric: 1350ms

Appending the first group key on the second fabric: 130ms (nearly double the first fabric).
Appending the first group key on the third fabric: 260ms
I have never seen the TC reach the 4th or 5th fabric (even with the timeout change).

Time to process the first 40 group keys on the first fabric:
      `Long dispatch time: 7428 ms, for event type 3`
Time to process the last 40 group keys (280-320) on the first fabric 
     `Long dispatch time: 44669 ms, for event type 3`

See the DUT logs for more data.

Even with the drastically increased `interactionTimeoutMS`, the step fails. The attached log also shows errors that are caused by processing the group key request for so long:
    `UnauthenticatedSession exhausted`
    `Failed to queue CHIP System Layer lambda event: 1000000`
    `CASESession timed out`

The append operation uses `GroupDataProviderImpl::SetGroupKeyAt()`.  It is not optimized to handle appending large numbers of group keys. There are possibilities for improvement - reusing references instead of loading everything repeatedly, or delay saving to nonvolatile storage until the end, etc. I have not measured the execution time of lines within `SetGroupKeyAt()`.

[DUT Logs extra logging.txt](https://github.com/user-attachments/files/21473337/DUT.Logs.extra.logging.txt)

[TH logs with extended timeout.log](https://github.com/user-attachments/files/21473340/TH.logs.with.extended.timeout.log)

mikesolsen, 2025-07-29:
Thanks for taking a look @wqx6. I am testing with 80 bridged end points. TC-RR-1.1 creates the maximum possible number of groups: 80 group clusters * 4 groups per cluster * 5 fabrics = 1600 groups (and apparently 1600 group keys). I do not expect to see this in the real world - the major platforms don't seem to be using Matter groups at all.

The errors at the end regarding `Failed to queue` are not the concern.  If I run the test without modifying the scripts `interactionTimeoutMS`, the bridge's event queue has no problem (but the test case fails very quickly).  I have previously increased CONFIG_MAX_EVENT_QUEUE_SIZE to 40 as per the consensus in other tickets here.

Instead, the concern is the inability of `WriteGroupKeyMap()` to handle the large number of group keys written by TC-DD-1.1. The dramatically increasing processing time from one key to another illustrates that the current approach is flawed.

To fix this:
1) Improve WriteGroupKeyMap(). The current approach has nice separation of concerns, but the performance is very poor for a bridge application.  Loading, parsing, and saving each element in the request is not efficient.
2) The requests sent by TC-RR-1.1 do not represent real world usage. Nobody creates 1600 groups in 5 requests. Modify TC-RR-1.1 to create group keys and groups in a series of reasonable requests. The current approach was likely written with a light bulb in mind, not a bridge.

I suspect both approaches will be required for step 14 to succeed. I suspect I will also have problems with step 15 of TC-RR-1.1: creating the groups after the group keys have been created.

wqx6, 2025-07-30:
The group key number does not increase when you add more application endpoints that include the Group cluster. What changes is the Group Key Map(the mapping between Group IDs and Group Key Set IDs), and this map grows as you add more endpoints with a Group cluster.

In my test with a device with 15 application endpoints (each having a Group cluster), I observed retransmission during Step 14 and Step 15, because the payload for the Group Key Map write/read interaction becomes quite large. This retransmission behavior is expected, and I did not encounter any queue exhaustion on the end device 

Since the Test Harness will only send the next chunked message after the DUT has handled the previous one and returned a report, performance degradation in WriteGroupKeyMap() does not break the flow. The only point that matters is the verification logic in TC_RR_1_1, which reads the number of groups (based on endpoints) and compares it to MaxGroupsPerFabric. For certificate test, you don't need to create so many bridged endpoints with group cluster, so it is OK.

In real-world use, having 300 group key maps (e.g. 15 endpoints × 4 groups per endpoint × 5 fabrics) is quite extreme. That's more of a stress test scenario than a typical certificate test.

Note: I had to increase the NVS partition size to 0x30000 in order to pass the test case.

code-overload42, 2026-09-30:
Hello @wqx6 ,

We face exactly the same problem and it is a certification blocker for us too. Are you checking this issue or working on a root fix for this ?

@crivi30 Did you find a solution or an arrangement with the test agency ?