espressif/esp-matter #1509: esp-matter attribute system has many problems with hybrid endpoints.

ehsanziyaee, 2025-07-31:
Suppose an endpoint—such as a microwave endpoint—contains multiple clusters, including Descriptor, Operational State, Microwave Mode, and Microwave Oven Control. Each of these clusters requires a delegate class that inherits from the corresponding cluster's delegate class in the CHIP framework.

Why should these be implemented in separate classes? Because each cluster within the endpoint must be assigned a distinct delegate instance.

Ideally—similar to the attribute system in esp-matter —each delegate class should be able to access or modify the attributes of other clusters within the same endpoint. This should be possible since attributes can be accessed using the endpoint ID and attribute ID.

However, this interoperability does not occur in practice. When reimplementing the delegate classes derived from the CHIP cluster interfaces, you do not gain access to other clusters’ attributes, as you might expect in an esp-matter-like environment.

Furthermore, in your callback implementation (e.g., MicrowaveOvenControlDelegateInitCB), you create two instances of each cluster under the endpoint. This duplication leads to additional issues, which I have discussed in a previous issue.

jadhavrohit924, 2025-08-04:
The core reason delegates are implemented by the application and not the SDK is their purpose: they are application-specific drivers. Delegates are not like attributes, they are interface classes designed for the application to implement its own unique logic. For a microwave, this logic might include how to handle state changes, power settings, or mode selections.

>However, this interoperability does not occur in practice. When reimplementing the delegate classes derived from the CHIP cluster interfaces, you do not gain access to other clusters’ attributes, as you might expect in an esp-matter-like environment.

Furthermore, delegates are intended to be self-contained drivers for a specific cluster, not a shared data access layer for the entire endpoint.

> Furthermore, in your callback implementation (e.g., MicrowaveOvenControlDelegateInitCB), you create two instances of each cluster under the endpoint. This duplication leads to additional issues, which I have discussed in this (https://github.com/espressif/esp-matter/issues/1486) issue.

This has been fixed.

ehsanziyaee, 2025-08-04:
@jadhavrohit924 The last commit and merge on the main branch show that this issue has been resolved.

dhrishi, 2025-08-25:
Closing this now