espressif/esp-matter #1517: Code to create concentration measurement cluster?

olavt, 2025-08-12:
Is this the correct code to create a concentration measurement cluster, or can the code be simoplified?

```cpp
    cluster::carbon_dioxide_concentration_measurement::config_t cluster_config;
    cluster_config.feature_flags =
        cluster::concentration_measurement::feature::numeric_measurement::get_id() |
        cluster::concentration_measurement::feature::average_measurement::get_id() |
        cluster::concentration_measurement::feature::peak_measurement::get_id();
    cluster_config.measurement_medium = static_cast<uint8_t>(CarbonDioxideConcentrationMeasurement::MeasurementMediumEnum::kAir);
    cluster_t* cluster = esp_matter::cluster::carbon_dioxide_concentration_measurement::create(m_endpoint, &cluster_config, CLUSTER_FLAG_SERVER);
    
    // Add the NumericMeasurement (MEA) Feature flag    
    cluster::carbon_dioxide_concentration_measurement::feature::numeric_measurement::config_t numeric_measurement_config;
    numeric_measurement_config.measurement_unit = static_cast<uint8_t>(CarbonDioxideConcentrationMeasurement::MeasurementUnitEnum::kPpm);
    cluster::carbon_dioxide_concentration_measurement::feature::numeric_measurement::add(cluster, &numeric_measurement_config);

    // Add the AverageMeasurement (AVG) Feature flag
    cluster::carbon_dioxide_concentration_measurement::feature::average_measurement::config_t average_measurement_config;
    average_measurement_config.average_measured_value_window = m_measurements.GetAverageWindowSizeSeconds(CarbonDioxideConcentrationMeasurement::Id);
    cluster::carbon_dioxide_concentration_measurement::feature::average_measurement::add(cluster, &average_measurement_config);

    // Add the PeakMeasurement (PEA) Feature flag
    cluster::carbon_dioxide_concentration_measurement::feature::peak_measurement::config_t peak_measurement_config;
    peak_measurement_config.peak_measured_value_window = m_measurements.GetPeakWindowSizeSeconds(CarbonDioxideConcentrationMeasurement::Id);
    cluster::carbon_dioxide_concentration_measurement::feature::peak_measurement::add(cluster, &peak_measurement_config);
```

It feels a bit redundant to first specify the feature flags and then add them again in order to specify some config for each feature (which I don't think can be passed in the config when creating the cluster.

I also get these warnings when adding the features, since the attributes already was created when creating the cluster:

W (670) data_model: Attribute 0x00000000 on cluster 0x0000040D already exists. Not creating again.
W (670) data_model: Attribute 0x00000001 on cluster 0x0000040D already exists. Not creating again.
W (670) data_model: Attribute 0x00000002 on cluster 0x0000040D already exists. Not creating again.
W (680) data_model: Attribute 0x00000008 on cluster 0x0000040D already exists. Not creating again.

I think this should be cleaned up somehow.

olavt, 2025-08-13:
@shubhamdp I'm using ec9699be7c8e5ff43420107e7236c7a33953b452.

It should be newer than the major refactoring that was done.

shubhamdp, 2025-09-03:
@olavt we have merged changes related to this in 519a3d0dc39e9e22b236f223006b1d13bea15069. Please pull the latest changes and give it a try.

It adds the features provided in the feature flags, and no need to using `::add()` on such clusters which has `O.a` or `O.a+` feature conformance.

Simplified code would look like
```
using namespace esp_matter::cluster::carbon_dioxide_concentration_measurement;

config_t cluster_config;
cluster_config.measurement_medium = static_cast<uint8_t>(CarbonDioxideConcentrationMeasurement::MeasurementMediumEnum::kAir);

// configure feature map
cluster_config.feature_flags = feature::numeric_measurement::get_id() |
        feature::average_measurement::get_id() |
        feature::peak_measurement::get_id();

// override any config values for features
cluster_config.features.numeric_measurement.measurement_unit = static_cast<uint8_t>(CarbonDioxideConcentrationMeasurement::MeasurementUnitEnum::kPpm);
cluster_config.features.peak_measurement.peak_measured_value_window = m_measurements.GetPeakWindowSizeSeconds(CarbonDioxideConcentrationMeasurement::Id);
cluster_config.features.average_measurement.average_measured_value_window = m_measurements.GetAverageWindowSizeSeconds(CarbonDioxideConcentrationMeasurement::Id);

cluster_t* cluster = create(m_endpoint, &cluster_config, CLUSTER_FLAG_SERVER);
```

Let me know if this works.

olavt, 2025-09-05:
@shubhamdp Perfect! Thanks!

shubhamdp, 2025-09-10:
Got it, even specification has same base cluster with only change in cluster-id.

Can you check if https://github.com/espressif/esp-matter/commit/c54786215109b2ec61a9d767a3f94b9ec1be9af0 works for you?

I have tested with this driver code https://github.com/espressif/esp-matter/commit/583e4cd5915a6963258dc06b4ed4079ae9680189

You can patch your main using following command
```
curl https://github.com/espressif/esp-matter/commit/c54786215109b2ec61a9d767a3f94b9ec1be9af0.patch | git am
```

re-build the example and you should be good to go.

Let me know if this works for you.

olavt, 2025-09-10:
@shubhamdp 

I think the changes look good.

One feedback:

I would rather have this "create_cm_cluster" renamed to "create_concentration_measurement_cluster" as it is more inline with the naming.

What about this enum:

CarbonDioxideConcentrationMeasurement::MeasurementMediumEnum::kAir

Is there a ConcentrationMeasurement::MeasurementMediumEnum::kAir constant available?

shubhamdp, 2025-09-10:
> I would rather have this "create_cm_cluster" renamed to "create_concentration_measurement_cluster" as it is more inline with the naming.

Just a driver code to test out changes, they won't go into the SDK.

```
CarbonDioxideConcentrationMeasurement::MeasurementMediumEnum::kAir
```

Looks like someone has handwritten these Enums here: https://github.com/espressif/connectedhomeip/blob/bdc38cd772964c67b43060cc399a2fe531529c39/src/app/clusters/concentration-measurement-server/concentration-measurement-cluster-objects.h

olavt, 2025-09-10:
@shubhamdp Yes, I was a bit to quick.

I see that the cluster create is as expected here:

esp_matter::cluster::concentration_measurement::create(endpoint, &cluster_config, CLUSTER_FLAG_SERVER, cluster_id);

Looks fine then!

When will it be in the main branch?

shubhamdp, 2025-09-29:
@olavt we have recently merged the above patch, please check https://github.com/espressif/esp-matter/commit/e264ecb0f15c72ea6fe6b12e0df2969d779c1fd2 and see if this is what you are looking for. And sorry for the delay!

olavt, 2025-09-29:
@shubhamdp Looks good!

shubhamdp, 2025-09-30:
@olavt Thanks for verifying. Can you please close the issue if all the concerns are resolved!

closed 2025-10-01 by olavt