espressif/esp-matter #1521: Add more appliance Controller (e.g. Oven Controller)

ehsanziyaee, 2025-08-20:
As you know, to send commands to appliances, you need to use their controller cluster APIs (via chip-tool commands and the corresponding callbacks). Currently, only a limited number of controllers are available for appliances (such as microwave controllers and laundry machine controllers). Does this mean we must wait for new appliances to have their controllers fully implemented in Matter/CHIP?

Note that this question refers to devices that already have a defined device type ID in Matter (e.g., an oven) but do not yet have an associated controller, and not to appliances that lack a standard device type ID (e.g., a coffee maker).

dhrishi, 2025-08-21:
@ehsanziyaee You do not need a specific controller device type to control that device type over Matter. Every device type is created from different clusters. Every cluster has a server and a client. A client cluster can send commands/requests to the corresponding server cluster. If your phone app/controller can instantiate the corresponding client clusters, you can interact with the server clusters, without needing the specific controller device type.

> or can we just use the clusters directly?

So, yes, this is right. Also, the specific controller device types (e.g. pump controller) in the spec are Matter end devices that can be commissioned and can interact with those device types (e.g. pump) through binding.

For your specific example, the oven device type in the spec is a model that has atmost one cooktop and minimum one temperature controlled cabinet. For your requirements, there's a device type microwave oven. This has Microwave Oven Control cluster as mandatory (1.4.1 device type library - section 13.11.6). You can use the command SetCookingParameters (1.4.1 application library - section 8.13.6.2) to set the cooking time. From your phone app/controller, you can use the Microwave Oven Control client cluster to achieve this.

ehsanziyaee, 2025-08-21:
@dhrishi Is it possible to use another cluster for cook time ? e.g., OnOff cluster (off time attribute), and placed it in our appliance model?

ehsanziyaee, 2025-09-06:
@dhrishi  I registered an Issue in the CHIP repo.
https://github.com/project-chip/connectedhomeip/issues/40746

dhrishi, 2025-09-13:
Ok. Thanks. I will close the issue here assuming you have/will be raising it in the spec repo for further discussion

closed 2025-09-13 by dhrishi