espressif/esp-matter #1532: Problem pairing Matter-over-Thread Light Bulbs with the esp-matter Controller example

tantotea, 2025-09-17:
### Issue
While testing pairings Matter-over-Thread devices with the Controller example, the pairings are successful EXCEPT on light bulbs (Tried on Nanoleaf and Aqara, both of which are Matter-over-Thread). 

But those light bulbs can be paired successfully when using:
- `$ chip-tool pairing code-thread <node-id> hex:<dataset> <setup-code>`
- Apple Home

### Environment
- ESP-Matter Commit Id: ca3df4f
- ESP-IDF Commit Id: esp-idf 5.4.2
- SoC: ESP32-S3+H2 (https://openthread.io/guides/border-router/espressif-esp32)

### Steps
1. Build the examples/controller, flash and monitor on Esp32-S3+H2
2. Set the Matter-over-Thread light bulbs (Nanoleaf or Aqara) in pairing mode
3. Use console to pair those light bulbs:
`> matter esp controller pairing code-thread (myNodeId) (threadDataset) (bulbSetupCode) 
`

### Log
```
I (308113) chip[CTL]: Setting thread operational dataset from parameters
I (308123) chip[CTL]: Setting attestation nonce to random value
I (308123) chip[CTL]: Setting CSR nonce to random value
I (308123) chip[CTL]: Checking ICD registration parameters
I (308123) chip[CTL]: Stopping commissionable node discovery over DNS-SD
I (308123) chip[CTL]: Starting commissionable node discovery over BLE
I (308123) chip[CTL]: Skipping commissionable node discovery over Wi-Fi PAF since not supported by the controller!
I (308123) chip[CTL]: Starting commissionable node discovery over DNS-SD
Done
I (308133) NimBLE: GAP procedure initiated: discovery; 
I (308133) NimBLE: own_addr_type=1 filter_policy=0 passive=1 limited=0 filter_duplicates=1                                                                                                                                                                                            
I (308133) NimBLE: duration=60000ms
I (308133) NimBLE: 

I (308143) chip[BLE]: Device Discriminator match. Attempting to connect
I (308143) chip[BLE]: Scan complete notification without an active scan
I (308143) NimBLE: GAP procedure initiated: connect; 
I (308143) NimBLE: peer_addr_type=1 peer_addr=
I (308143) NimBLE: 7f:d4:3c:d3:f7:c5
I (308143) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=1
I (308153) NimBLE: 

I (308343) NimBLE: GATT procedure initiated: exchange mtu

E (308343) chip[DL]: Disabling CHIPoBLE service due to error: ac
I (308343) NimBLE: GAP procedure initiated: stop advertising.

I (308343) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (308443) chip[DL]: BLE GAP connection terminated (con 1 reason 0x216)
I (308443) chip[DL]: HandlePlatformSpecificBLEEvent 32775
E (328143) chip[BLE]: Failed to cancel connection rc=30
I (328143) chip[DL]: HandlePlatformSpecificBLEEvent 16386
E (328143) chip[CTL]: Commissionable node discovery over BLE failed: 32
E (328143) chip[-]: Error CHIP:0x00000032 at src/controller/SetUpCodePairer.cpp:442
E (338133) chip[CTL]: Discovery timed out
I (338133) chip[CTL]: Stopping commissionable node discovery over DNS-SD
```

But those light bulbs can be paired successfully when using chip-tool and Apple Home.

### Expected Results
It is expected that pairing the Matter-over-Thread light bulbs is successful using the Controller on Esp32-S3+H2.

tantotea, 2025-09-18:
Is there perhaps some configs that need to be set for pairing the light bulbs?  The controller works for pairing other matter devices though.

wqx6, 2026-10-23:
https://github.com/project-chip/connectedhomeip/pull/74922 This PR might help, @tantotea @xbln Could you try to commissioning with this fix?