espressif/esp-matter #1534: Alexa is no recognising well all my clusters / endpoints

aroldan-SP, 2025-09-18:
I'm trying to develop a test product of an air purifier and also an air quality sensor (two different products/code) bu tI always got some "malfunctions" with the controller. All the commisioning process and control of the product works perfectly fine, but for example, in my air purifier app i have this schema:

- Endpoint 1: Air purifier (0x002D)
    - FanControl Cluster (M - 0x0202)
    - On/Off Cluster (O - 0x0006)
    - HEPA Filter Monitoring Cluster (O - 0x0071)
- Endpoint 2: Temperature Sensor (0x0302) (Child of endpoint 1)
- Endpoint 3: Humidity Sensor (0x0307) (Child of endpoint 1)

Here is my code:

`
    //-------------------------------------------------------
    //---------------- Air Purifier Endpoint ----------------
    //-------------------------------------------------------
    esp_matter::endpoint::air_purifier::config_t air_purifier_config;
    endpoint_t *air_purifier_endpoint = air_purifier::create(node, &air_purifier_config, ENDPOINT_FLAG_NONE, NULL);
    ABORT_APP_ON_FAILURE(air_purifier_endpoint != nullptr, ESP_LOGE(TAG, "Failed to create air purifier endpoint"));

    //-------------------------------------------------------
    //----------------- Fan Control Cluster -----------------
    //-------------------------------------------------------
    // cluster_t *fan_control_cluster = cluster::get(air_purifier_endpoint, chip::app::Clusters::FanControl::Id);

    // Fan mode auto enabled
    // cluster::fan_control::feature::fan_auto::add(fan_control_cluster);

    // Airflow direction feature
    // esp_matter::cluster::fan_control::feature::airflow_direction::config_t airflow_direction_config;
    // airflow_direction_config.airflow_direction = static_cast<uint8_t>(FanControl::AirflowDirectionEnum::kForward);
    // cluster::fan_control::feature::airflow_direction::add(fan_control_cluster, &airflow_direction_config); // O

    // Config mode //////// Test Nº1
    esp_matter::cluster::fan_control::config_t fan_control_config;
    fan_control_config.fan_mode = static_cast<uint8_t>(FanControl::FanModeEnum::kOff);
    fan_control_config.fan_mode_sequence = static_cast<uint8_t>(FanControl::FanModeSequenceEnum::kOffLowMedHighAuto);
    fan_control_config.percent_setting = (uint8_t) 0;
    fan_control_config.percent_current = 0;
    cluster_t *fan_control_cluster = cluster::fan_control::create(air_purifier_endpoint, &fan_control_config, CLUSTER_FLAG_SERVER);
    cluster::fan_control::feature::fan_auto::add(fan_control_cluster);

    // esp_matter SDK functions config //////// Test Nº2
    // fan_control::attribute::create_fan_mode(fan_control_cluster, static_cast<uint8_t>(FanControl::FanModeEnum::kOff)); // M
    // fan_control::attribute::create_fan_mode_sequence(fan_control_cluster, static_cast<uint8_t>(FanControl::FanModeSequenceEnum::kOffLowMedHighAuto)); // M
    // fan_control::attribute::create_percent_setting(fan_control_cluster, 0); // M
    // fan_control::attribute::create_percent_current(fan_control_cluster, 0); // M

    //-------------------------------------------------------
    //-------------------- On_Off Cluster -------------------
    //-------------------------------------------------------
    esp_matter::cluster::on_off::config_t on_off_config;
    cluster_t *on_off_cluster = cluster::on_off::create(air_purifier_endpoint, &on_off_config, CLUSTER_FLAG_SERVER);

    //-------------------------------------------------------
    //------------ HEPA Filter Monitoring Cluster -----------
    //-------------------------------------------------------
    esp_matter::cluster::hepa_filter_monitoring::config_t hepa_filter_config;
    cluster_t *hepa_filter_cluster = cluster::hepa_filter_monitoring::create(air_purifier_endpoint, &hepa_filter_config, CLUSTER_FLAG_SERVER);
    hepa_filter_monitoring::attribute::create_change_indication(hepa_filter_cluster, static_cast<uint8_t>(HepaFilterMonitoring::ChangeIndicationEnum::kOk)); // M

    // Add condition feature //////// Test Nº1
    cluster::hepa_filter_monitoring::feature::condition::config_t hepa_filter_condition_config;
    hepa_filter_condition_config.condition = 100;
    hepa_filter_condition_config.degradation_direction = static_cast<uint8_t>(HepaFilterMonitoring::DegradationDirectionEnum::kDown);
    cluster::hepa_filter_monitoring::feature::condition::add(hepa_filter_cluster, &hepa_filter_condition_config);
    // Add condition and degradation direction directly as attributes //////// Test Nº2
    // hepa_filter_monitoring::attribute::create_condition(hepa_filter_cluster, 100); // O
    // hepa_filter_monitoring::attribute::create_degradation_direction(hepa_filter_cluster, static_cast<uint8_t>(HepaFilterMonitoring::DegradationDirectionEnum::kDown)); // O
    
    //-------------------------------------------------------
[... 38 lines cut ...]
    - PM2.5 Concentration Measurement Cluster (O - 0x042A)
    - Total Volatile Organic Compounds Concentration Measurement Cluster (O - 0x042F)

With all this setup the expected Alexa outcome would be something like this:

[image]

But in my case is like this:

[image]

[image]

I'm losing it trying to find what I am doing wrong, I've tried lots of diferent configurations and seeked through the documentation and also debuged with the CHIP tool everything, but all seems to be correct except in the controller app. It could be something with the controller? What I am missing?

Would love any comment trying to solve this. Thanks. 

- ESP-Matter Commit Id: [db891d9](https://github.com/espressif/esp-matter/commit/db891d9f4e70fad83fd01d890f77c1625cd41f0d)
- ESP-IDF Commit Id: [8c750b0](https://github.com/espressif/esp-idf/commit/8c750b088c7cd857d079c0eeb495da199b359461)
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3

- Host Machine OS: Ubuntu 24.04.3 LTS
- Host Machine Python version: 3.12.3

- Commissioner app and versions if present: Echo Pop with software version 12383141252

jadhavrohit924, 2025-09-22:
@aroldan-SP Your data model looks correct, we have observed same behaviour on Alexa. However we have tried this on apple home as well, where are can see 3 endpoints correctly. We suspect this could be an issue with the Alexa.

aroldan-SP, 2025-09-22:
@jadhavrohit924 Ok thanks for it, I'll try to test with another controller to test if it works better. Could you give me some advice on Which controller should I pick for testing? Wich one could have less compatibility Issues like this? I thought about Google home or as you tested apple home.

Thank you!

jadhavrohit924, 2025-11-11:
We generally prefer Google home and Apple home.

closed 2025-11-11 by aroldan-SP