espressif/esp-matter #1537: vscode 编译不通过

dading105, 2025-09-21:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:db891d9
- ESP-IDF Commit Id:5.4.1
- SoC (eg: ESP32 or ESP32-C6):ESP32-C6
- Device Logs (Please attach the log file):

- Host Machine OS:   windows
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**

 *  Executing task: C:\Users\dadin\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe d:\Develop\esp\v5.4.1\esp-idf\tools\idf.py -B e:\aitk\imu\esp-matter\examples\light\build -DSDKCONFIG='e:\aitk\imu\esp-matter\examples\light\sdkconfig' reconfigure 

Executing action: reconfigure
Running cmake in directory E:\aitk\imu\esp-matter\examples\light\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\dadin\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DSDKCONFIG='e:\aitk\imu\esp-matter\examples\light\sdkconfig' -DCCACHE_ENABLE=1 E:\aitk\imu\esp-matter\examples\light"...
-- Found Git: C:/Users/dadin/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
IDF_VERSION: 5.4.1; idf version number: 5.4.1
Sequence of SDKCONFIG files: sdkconfig.defaults
-- Component directory E:/aitk/imu/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory E:/aitk/imu/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory E:/aitk/imu/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory E:/aitk/imu/esp-matter/examples/common/manufacturing_data_provider does not contain a CMakeLists.txt file. No component will be added
-- Component directory E:/aitk/imu/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory E:/aitk/imu/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
CMake Error at D:/Develop/esp/v5.4.1/esp-idf/tools/cmake/project.cmake:472 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  E:/aitk/imu/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components   
Call Stack (most recent call first):
  D:/Develop/esp/v5.4.1/esp-idf/tools/cmake/project.cmake:556 (__project_init)     
  CMakeLists.txt:49 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the E:\aitk\imu\esp-matter\examples\light\build\log\idf_py_stderr_output_19708 and E:\aitk\imu\esp-matter\examples\light\build\log\idf_py_stdout_output_19708

 *  The terminal process "C:\Users\dadin\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe 'd:\Develop\esp\v5.4.1\esp-idf\tools\idf.py', '-B', 'e:\aitk\imu\esp-matter\examples\light\build', '-DSDKCONFIG='e:\aitk\imu\esp-matter\examples\light\sdkconfig'', 'reconfigure'" terminated with exit code: 2.

dading105, 2025-09-22:
有人帮忙看下？

wqx6, 2025-09-22:
由于esp-matter需要使用gn，所以目前SDK并不支持windows，你可以选择使用WSL或者是虚拟机来开发matter应用，关于如何在使用WSL以及VScode开发matter应用，请参考https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10

dading105, 2025-09-22:
@wqx6 谢谢

closed 2025-11-06 by dhrishi