espressif/esp-matter #1543: 安装install_packages(gn_target)报错

dading105, 2025-09-25:
install_packages(gn_target)
  File "/home/dt/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 407, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/dt/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/dt/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

./install.sh运行报错

dading105, 2025-09-26:
哪位大佬看下！

dading105, 2025-09-27:
ubuntu20.04好像不行

Jerry-ESP, 2025-09-28:
When you setup the latest esp-matter, please use ubuntu 24.04 , and use a vpn when you execute install.sh

closed 2025-11-06 by dhrishi